aurora-opensource/streetscape.gl #552: Metric/Plot Data zoom in/out for analysis

ShekharSumanP, 2022-08-12:
Hi
Thanks for the amazing toolkit. 

I wanted to get some suggestions/idea for the metric/plot visualisation capable of zooming in and out. From what i have read such feature is not available at the moment.

The point here being that currently we are not able to zoom into a particular time -series range within the whole dataset, something we can do using [plotjuggler](https://www.plotjuggler.io/) or Grafana. This could come in real handy in sudden random data points for example jerks or high freq oscillations for debugging/analysis of our system.

Would really appreciate any ideas/comments on this.

Regards
Shekhar Suman Patel

yujingz, 2022-08-31:
We've had similar stuff done with Chart.js and it's zoomable plugin https://www.chartjs.org/chartjs-plugin-zoom/latest/guide/usage.html

At the end of the day, the plot component is just a presentation layer. You can show the data in whatever react component you want.

ShekharSumanP, 2022-09-09:
Thank you so much for the answer.

closed 2022-09-09 by ShekharSumanP