aurora-opensource/streetscape.gl #560: When I switch from the DRIVER visual angle to the PERSPECTIVE/TOPDOWN visual angle, the screen will have problems

heyu3913, 2022-11-01:
PERSPECTIVE visual angle
[image]
TOP DOWN visual angle
[image]

Nothing changed

twojtasz, 2022-11-01:
What problems? I see the bottom of the polygons in perspective, and don't
in top down orthogonal which is correct.

What is the driver part of this?

heyu3913, 2022-11-02:
Thank you so much for your reply. 
Here are my operation steps：

Step1. The default click to open the page is PERSPECTIVE MODE, the page is shown as below：
[image]

Step2. When I switch to TOP-DOWN MODE, the page is shown as below：
[image]

Step3. When I switch to DRIVE MODE, the page is shown as below：
[image]

Step4. When I switch to PERSPECTIVE -> TOP-DOWN MODE again, the page is shown as below：
[image]

My questions as below：
1. Why the page was different from step2 when i did the step4 switching back to TOP-DOWN MODE？
2. How can i return to the page of step1 when i am in step3？
3. I could change the angle of view by my mouse and keyboard when i was in step1 and step2, but I couldn’t do in step3 or step4.

Looking forward to your advice.

twojtasz, 2022-11-04:
Did you find the fix?

twojtasz, 2022-11-04:
Seems like transition in step 2 is wrong, but in First person mode controls
are limited, same with top down.  Just rotation in those modes

heyu3913, 2022-11-04:
I didn't fix it

PabloVD, 2025-01-29:
Hey all, this issue still persists.